zephyrproject-rtos/test_results #656: tests-ci : drivers: disk: usdhc test failed

hakehuang, 2021-12-23:
**Describe the bug**
usdhc test is failed on v2.7.99-2628-g38ce8d615d4b on mimxrt1050_evk

see logs for details

**To Reproduce**
1. 
```
scripts/twister --device-testing --device-serial /dev/ttyACM0 -p mimxrt1050_evk  --sub-test drivers.disk
```
2. See error

**Expected behavior**
test pass

**Impact**

**Logs and console output**
```
*** Booting Zephyr OS build v2.7.99-2628-g38ce8d615d4b  ***
Running test suite disk_driver_test
===================================================================
START - test_setup
E: NO SD inserted!
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:50: test_setup: (rc not equal to 0)
Disk access initialization failed
FAIL - test_setup in 0.115 seconds
===================================================================
START - test_read
Testing reads of 8 sectors
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:95: test_sector_read: (rc not equal to 0)
Failed to read from sector zero
FAIL - test_read in 0.16 seconds
===================================================================
START - test_write
Testing writes of 8 sectors
Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/disk/src/main.c:148: test_sector_write: (rc not equal to 0)
Failed to write to sector zero
FAIL - test_write in 0.16 seconds
===================================================================
Test suite disk_driver_test failed.
===================================================================
PROJECT EXECUTION FAILED

    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: v2.7.99-2628-g38ce8d615d4b

hakehuang, 2021-12-23:
Also fails on mimxrt1050_evk for v2.7.99-2628-g38ce8d615d4b

hakehuang, 2021-12-23:
@chen-png do you connect sd card for this case?

chen-png, 2021-12-23:
>  do you connect sd card for this case?

No, it also failed in our daily test results.

hakehuang, 2021-12-23:
> No, it also failed in our daily test results.

it is created from your daily test report, and for this case, you need insert SD card

chen-png, 2021-12-23:
thanks, I will test it again with a SD card.

hakehuang, 2021-12-24:
@chen-png thanks let me know whether it works for you.

hakehuang, 2022-01-04:
need insert sd card for testing pass

closed 2022-01-04 by hakehuang

chen-png, 2022-01-06:
sd card is still on the way, will test this when I got it.

hakehuang, 2022-01-06:
Thanks for updating. @chen-png

chen-png, 2022-01-07:
after inserting a sd card on mimxrt1050_evk board, this issue was fixed.

>*** Booting Zephyr OS build v2.7.99-2908-ge2588d6a4293  ***
Running test suite disk_driver_test
===================================================================
START - test_setup
I: SD inserted!
Disk reports 121634816 sectors
Disk reports sector size 512
 PASS - test_setup in 0.112 seconds
===================================================================
START - test_read
Testing reads of 8 sectors
Testing reads of 1 sectors
Testing reads of 29 sectors
Testing reads of 31 sectors
 PASS - test_read in 0.19 seconds
===================================================================
START - test_write
Testing writes of 8 sectors
Testing writes of 1 sectors
Testing writes of 29 sectors
Testing writes of 31 sectors
 PASS - test_write in 0.244 seconds
===================================================================
Test suite disk_driver_test succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL