zephyrproject-rtos/test_results #935: TCP Close This test changes FIN/ACK segments into SYN/ACK segments. error

hakehuang, 2022-04-06:
**Describe the bug**
This test changes FIN/ACK segments into SYN/ACK segments. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: section 3.9

**Results**

FAIL: tcp.v4 did not see expected RST from DUT. Returned "Echo response", which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2022-07-29:
This looks like a valid issue, created a Zephyr issue to track progress: https://github.com/zephyrproject-rtos/zephyr/issues/48465.

ssharks, 2023-01-06:
@hakehuang : Could you recheck this issues at I think it has been solved by @rlubos

hakehuang, 2023-01-08:
@ssharks confirmed

closed 2023-01-08 by hakehuang