zephyrproject-rtos/test_results #936: TCP Close Ignore data in RST segments. error

hakehuang, 2022-04-06:
**Describe the bug**
Ignore data in RST segments. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 1122: section 4.2.2.12 {RST can contain data}

**Results**

FAIL: tcp.v4 got echo response, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2022-07-29:
@hakehuang Can you provide a Wireshark pcap for this one, I'd like to see what's going on. Also, it's not clear to me what is the expected pass scenario for this test? Currently we just drop the RST packet and abort the connection. The linked RFC is also not clear about the RST segment containing data:
```
         4.2.2.12  RST Segment: RFC-793 Section 3.4

            A TCP SHOULD allow a received RST segment to include data.

            DISCUSSION
                 It has been suggested that a RST segment could contain
                 ASCII text that encoded and explained the cause of the
                 RST.  No standard has yet been established for such
                 data.
```

hakehuang, 2022-08-01:
[Uploading tcp_close_ignore_data_in_rst.zip…]()
@rlubos  please check the attached pcap

rlubos, 2022-08-01:
> Uploading tcp_close_ignore_data_in_rst.zip… @rlubos please check the attached pcap

It seems that the file did not upload properly?

hakehuang, 2022-08-02:
[tcp_close_ignore_data_in_rst.zip](https://github.com/zephyrproject-rtos/test_results/files/9238614/tcp_close_ignore_data_in_rst.zip)
reupload, @rlubos

rlubos, 2022-08-02:
Hmm, so the logs from Wireshark doesn't seem to follow the issue description:
> FAIL: tcp.v4 got echo response, which is not an expected result.

There is not reply from DUT in the logs. It's still not clear to me what is the cultprit of the failure here. One idea is that perhaps Zephyr should reply with ICMP Destination Unreachable in this case, but I'm not sure if that's the core problem here.

hakehuang, 2022-08-03:
looks like after server sends the close package, the DUT does not reply close status.

ssharks, 2023-02-07:
I thinks this test tries to validate the following section from the metioned RFC:
```
      fourth other text or control

        Any other control or text-bearing segment (not containing SYN)
        must have an ACK and thus would be discarded by the ACK
        processing.  An incoming RST segment could not be valid, since
        it could not have been sent in response to anything sent by this
        incarnation of the connection.  So you are unlikely to get here,
        but if you do, drop the segment, and return.
```

hakehuang, 2026-10-30:
Closing as part of the requested synchronization to `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`.

`TCP.Close.004` has no failing results in `test_results.csv`: all 2 reported results are Pass.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Pass | b'PASS: tcp.v4 got connection reset by peer on recv call, which is an expected result.' |
| `tcp.v6` | Pass | b'PASS: tcp.v6 got connection reset by peer on recv call, which is an expected result.' |

This closure reflects the supplied board's report; it does not claim that a historical platform such as qemu_x86 was retested.

closed 2026-10-30 by hakehuang