zephyrproject-rtos/test_results #938: TCP Connected Split TCP segments into 8 byte segments, set push, and send in reverse order. error

hakehuang, 2022-04-06:
**Describe the bug**
Split TCP segments into 8 byte segments, set push, and send in reverse order. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: section 3.9
RFC 1122: section 4.2.2.2 {Aggregate or queue un-pushed data}
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

FAIL: tcp.v4 got no echo response, which is not an expected result.; FAIL: tcp.v4 got no echo response, which is not an expected result.; FAIL: tcp.v6 got no echo response, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

ssharks, 2023-01-19:
@hakehuang: Could you retest this one? I think this one could have been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/53592 as well.

hakehuang, 2023-01-24:
pass on v3.2.0-3518-g7b5780e007

hakehuang, 2023-02-28:
test pass in v3.3.0-475-g38f554ef4f

hakehuang, 2026-10-30:
Reopening: `TCP.Connected.004` is failing again in the supplied report for `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`. The issue body now contains the current results; the original report is preserved as historical context. This does not establish that the earlier fix or historical board has regressed; the new failure is on the board named above.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Fail | b'FAIL: tcp.v4 got no echo response, which is not an expected result.' |
| `tcp.v6` | Fail | b'FAIL: tcp.v6 got no echo response, which is not an expected result.' |