zephyrproject-rtos/test_results #942: TCP Connected Segment: 2 bytes. Order: N ... 3 2 1. Overlap: None. error

hakehuang, 2022-04-06:
**Describe the bug**
Segment: 2 bytes. Order: N ... 3 2 1. Overlap: None. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: page 53
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

ssharks, 2023-01-06:
@hakehuang: Could you retest this one?

hakehuang, 2023-01-08:
@ssharks this issue still fails on zephyr-v3.2.0-3013-ge4fcb32451c5

hakehuang, 2023-01-19:
fixed in zephyr-v3.2.0-3518-g7b5780e007

hakehuang, 2026-10-30:
Reopening: `TCP.Connected.024` is failing again in the supplied report for `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`. The issue body now contains the current results; the original report is preserved as historical context. This does not establish that the earlier fix or historical board has regressed; the new failure is on the board named above.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Fail | b'FAIL: tcp.v4 got no echo response, which is not an expected result.' |
| `tcp.v6` | Fail | b'FAIL: tcp.v6 got no echo response, which is not an expected result.' |