zephyrproject-rtos/test_results #943: TCP Connected Segment: 3 bytes. Order: 1 2 3 ... N. Overlap: 1 byte. error

hakehuang, 2022-04-06:
**Describe the bug**
Segment: 3 bytes. Order: 1 2 3 ... N. Overlap: 1 byte. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: page 53
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

ssharks, 2023-01-06:
@hakehuang: Could you retest this one?

hakehuang, 2023-01-08:
@ssharks still fails on zephyr-v3.2.0-3013-ge4fcb32451c5

ssharks, 2023-01-08:
Interesting, especially as one situation seems to pass on IPv4, but fails with IPv6. In the summer I put in some effort to get this functionality working. This is under PR https://github.com/zephyrproject-rtos/zephyr/pull/47115. 

I'have put up a special testcase to test for these kind of issues, see https://github.com/zephyrproject-rtos/zephyr/blob/c3d39d95734c233e9ff2d1dbc2b6384fee9e8661/tests/net/tcp/src/main.c#L1397
There is a pretty elaborate list of nasty combinations that are being tested.

One of the key configuration values for the success is CONFIG_NET_TCP_RECV_QUEUE_TIMEOUT, it could be the default value of 100 is way to low. Could you have a look what value is used for your test and see what happens if you run the test with CONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=5000 in your prj.conf?

If this still does not work, could you help me to modify the list in https://github.com/zephyrproject-rtos/zephyr/blob/c3d39d95734c233e9ff2d1dbc2b6384fee9e8661/tests/net/tcp/src/main.c#L1375 to recreate the issue?

ssharks, 2023-01-08:
To run test test with the out of order testcases:
./scripts/twister -T tests/net/tcp -p qemu_x86

Just now I ran the more integration tests of TCP with the CONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=5000 in:
 ./scripts/twister -T tests/net/socket/tcp -p qemu_x86
 This worked just fine.

hakehuang, 2023-01-09:
@ssharks  do you think it is needed config? CONFIG_NET_TCP_RECV_QUEUE_TIMEOUT=5000

ssharks, 2023-01-09:
Yes. Could you check that? If so, i'll create a PR.

ssharks, 2023-01-09:
I've prepared the PR already https://github.com/zephyrproject-rtos/zephyr/pull/53592

ssharks, 2023-01-10:
The merge request has been merged, you if you want to re-test, you can take the current master as well

hakehuang, 2023-01-17:
@ssharks, it still fails at zephyr-v3.2.0-3518-g7b5780e007, and what we discuss here is to use 5000 as timeout, but why the PR sets to 2000?

ssharks, 2023-01-17:
Initially I submitted with 5 seconds, but it was found to be too long by the maintainers. So I reduced it to 2 seconds. Under normal circumstances that should be plenty to make the functionality work.

hakehuang, 2023-01-18:
> Initially I submitted with 5 seconds, but it was found to be too long by the maintainers. So I reduced it to 2 seconds. Under normal circumstances that should be plenty to make the functionality work.

I see, I will try to retest them maybe this is just because my test envirnment is slow.

hakehuang, 2023-01-19:
fixed with proper delay

hakehuang, 2026-10-30:
Reopening: `TCP.Connected.026` is failing again in the supplied report for `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`. The issue body now contains the current results; the original report is preserved as historical context. This does not establish that the earlier fix or historical board has regressed; the new failure is on the board named above.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Fail | b'FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.' |
| `tcp.v6` | Fail | b'FAIL: tcp.v6 got fewer bytes than expected, which is not an expected result.' |