zephyrproject-rtos/test_results #944: TCP Connected Segment: 6 segments. Order: Complex. Overlap: Complex. error

hakehuang, 2022-04-06:
**Describe the bug**
Segment: 6 segments. Order: Complex. Overlap: Complex. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: page 53
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

ssharks, 2022-07-29:
See https://github.com/zephyrproject-rtos/test_results/issues/948#issuecomment-1199208802

ssharks, 2023-01-06:
@hakehuang: Could you retest this one?

hakehuang, 2023-01-08:
@ssharks still fails on zephyr-v3.2.0-3013-ge4fcb32451c5