zephyrproject-rtos/test_results #946: TCP Connected Segment: 3 bytes. Order: N ... 3 2 1. Overlap: 1 byte. error

hakehuang, 2022-04-06:
**Describe the bug**
Segment: 3 bytes. Order: N ... 3 2 1. Overlap: 1 byte. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: page 53
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

FAIL: tcp.v4 got no echo response, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

ssharks, 2022-07-29:
See https://github.com/zephyrproject-rtos/test_results/issues/948#issuecomment-1199208802

ssharks, 2023-01-06:
@hakehuang: Could you retest this one?

hakehuang, 2023-01-08:
@ssharks still fails on zephyr-v3.2.0-3013-ge4fcb32451c5

hakehuang, 2023-01-19:
pass on zephyr-v3.2.0-3664-gad06dfc784

ssharks, 2023-01-19:
Thanks for testing again

hakehuang, 2026-10-30:
Reopening: `TCP.Connected.030` is failing again in the supplied report for `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`. The issue body now contains the current results; the original report is preserved as historical context. This does not establish that the earlier fix or historical board has regressed; the new failure is on the board named above.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Fail | b'FAIL: tcp.v4 got no echo response, which is not an expected result.' |
| `tcp.v6` | Fail | b'FAIL: tcp.v6 got no echo response, which is not an expected result.' |