zephyrproject-rtos/test_results #948: TCP Connected Segment: 6 segments. Order: Complex reversed. Overlap: Complex. error

hakehuang, 2022-04-06:
**Describe the bug**
Segment: 6 segments. Order: Complex reversed. Overlap: Complex. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 793: page 53
RFC 1122: section 4.2.2.20 {Process all Q'd before send ACK}
RFC 1122: section 4.2.2.20 {Queue out-of-order segments}
RFC 1122: section 4.2.2.21 {Send ACK for out-of-order segment}

**Results**

SKIPPED: tcp.v4: Packets exchanged with the DUT did not match those needed to either perform or grade the test. (Traffic source reports: no echo response.); FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.; FAIL: tcp.v6 got no echo response, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2022-07-29:
Fixes for out of order data processing are currently on a PR: https://github.com/zephyrproject-rtos/zephyr/pull/47115. Will need to re-test once merged.
CC @ssharks

ssharks, 2022-07-29:
It should resolve most issues, there is only one corner case remaining: When part of the incoming packet is already acknowledged. Even after this fix the packet will still be ignored.

ssharks, 2022-09-12:
PR has been merged

closed 2022-09-13 by hakehuang

ssharks, 2022-09-13:
Did you retest this one and the referred issues?