zephyrproject-rtos/test_results #950: TCP Connected Verify zero window probe occurs with increasing RTO. error

hakehuang, 2022-04-06:
**Describe the bug**
Verify zero window probe occurs with increasing RTO. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 1122: section 4.2.2.17 {First probe after RTO}
RFC 1122: section 4.2.2.17 {Exponential backoff}

**Results**

FAIL: tcp.v4. Timeouts periods seen: 

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2022-07-29:
ZWP sending was not implemented in Zephyr v3.0.0, it is present in Zephyr v3.1.0 though. Would be good to re-test this one. The question is, as we send Keep-alive packet (like Linux) as a ZWP, if it's acceptable by the test suite.

ssharks, 2022-07-29:
Exponential backoff is still not implemented for ZWP I think, also not in https://github.com/zephyrproject-rtos/zephyr/pull/46871

rlubos, 2022-07-29:
> Exponential backoff is still not implemented for ZWP I think, also not in [zephyrproject-rtos/zephyr#46871](https://github.com/zephyrproject-rtos/zephyr/pull/46871)

Good point, it should be moved Zephyr then.

ssharks, 2022-09-12:
PR has been merged.

hakehuang, 2026-10-30:
Reopening: `TCP.Connected.036` is failing again in the supplied report for `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`. The issue body now contains the current results; the original report is preserved as historical context. This does not establish that the earlier fix or historical board has regressed; the new failure is on the board named above.

| Traffic source | Result | Response |
|---|---|---|
| `tcp.v4` | Fail | b'FAIL: tcp.v4. Timeouts periods seen: 0.536432 0.916442 (1.7084/1.0 : BAD) 2.20839 (2.40975/1.0 : BAD) 3.77612 (1.70989/1.0 : BAD) 9.53607 (2.52536/1.0 : BAD) 15.9999 (1.67783/1.0 : BAD) 26.6233 (1.66396/1.0 : BAD) ' |
| `tcp.v6` | Fail | b'FAIL: tcp.v6. Timeouts periods seen: 0.468128 1.19242 (2.54721/1.0 : BAD) 2.00045 (1.67764/1.0 : BAD) 3.84026 (1.91969/1.0 : OK) 6.43206 (1.6749/1.0 : BAD) 15.4238 (2.39795/1.0 : BAD) ' |