zephyrproject-rtos/test_results #953: TCP Options Verify use of default TCP MSS value. error

hakehuang, 2022-04-06:
**Describe the bug**
Verify use of default TCP MSS value. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 1122: section 4.2.2.6 {Send-MSS default is 536}
RFC 1122: section 4.2.2.6 {Calculate effective send seg size}
RFC 6691

**Results**

SKIPPED: tcp.v4: Packets exchanged with the DUT did not match those needed to either perform or grade the test. (Traffic source reports: no echo response.); FAIL: tcp.v4 got 1040 bytes in a packet. Only 576 or fewer is expected.; SKIPPED: tcp.v6 got only 1000 total data bytes. Need at least 1280 to grade.; SKIPPED: tcp.v4: Packets exchanged with the DUT did not match those needed to either perform or grade the test. (Traffic source reports: no echo response.); SKIPPED: tcp.v6 got only 0 total data bytes. Need at least 576 to grade.; FAIL: tcp.v4 got 1040 bytes in a packet. Only 576 or fewer is expected.; SKIPPED: tcp.v6 got only 1000 total data bytes. Need at least 1280 to grade.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

ssharks, 2022-07-29:
Probably this issues has been fixed by commit: https://github.com/zephyrproject-rtos/zephyr/commit/f5679ab73e5dc3e3941e24a49c8a9e5d907a1d59
CC @rlubos

hakehuang, 2022-08-01:
@rlubos,  @ssharks this issues is still there see attached pcap
[Uploading tcp_options_verify_use_of_default_mss_value.zip…]()

tcp.v4 got 1040 bytes in a packet. Only 576 or fewer is expected

ssharks, 2022-08-01:
Thanks a lot for the testing. You are right, the default MTU is 1280 bytes, which should be 576 bytes for IPv4, this is still not fixed. The referred issue only fixed the case where the MSS was negotiated, but ignored in the transmission.

Interesting is that the testcode actually violates the spec itself in order to conduct the test...

rlubos, 2022-08-05:
Corresponding issue: https://github.com/zephyrproject-rtos/zephyr/issues/48739

ssharks, 2022-09-12:
PR has been merged

closed 2022-09-13 by hakehuang