zephyrproject-rtos/test_results #954: TCP Setup Set the unused field in Destination Unreachable to a non-zero value. error

hakehuang, 2022-04-06:
**Describe the bug**
Set the unused field in Destination Unreachable to a non-zero value. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 792: page 4
RFC 1122: section 3.2.2.1 {Pass ICMP Dest Unreachable to higher layer}
RFC 1122: section 4.2.3.9 {Dest. Unreach (2-4) => abort conn}
RFC 1122: section 4.2.3.9 {Receiving ICMP Messages from IP}

**Results**

PASS: tcp.v4 got no reply packets, which is an expected result.; FAIL: tcp.v4 got reply packets, which is not an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-06:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2022-07-29:
I wonder if this is false-positive, just checked with a dummy application, and Zephyr does set the "unused" field in the ICPMv4 packet to 0:
[image]
Perhaps we should check the Wireshark pcap from this test case?

hakehuang, 2022-08-01:
[tcp_setup_destination_unused.zip](https://github.com/zephyrproject-rtos/test_results/files/9230676/tcp_setup_destination_unused.zip)

@rlubos, the case is to set the destination to a unused value, please check attached pcap.

rlubos, 2022-08-01:
> [tcp_setup_destination_unused.zip](https://github.com/zephyrproject-rtos/test_results/files/9230676/tcp_setup_destination_unused.zip)
> 
> @rlubos, the case is to set the destination to a unused value, please check attached pcap.

I don't follow, what "desitnation" is modified in this pcap file?

hakehuang, 2022-08-02:
[image]

please check the red columnned package, test server send out the setup data to a non-zero value, and tccp shall not reply this.

ssharks, 2022-08-02:
@rlubos requested me to look at it as well, and I must admit I'm almost as puzzled as he is. If I understand you correctly you are testing some features of ICMP / TCP. The fail or passed is based on on a TCP connection returning data or not. 

In the Wireshark trace I do see a echo-reply on which I don't see anything unexpected at first glance. To me it looks like we are missing some pieces of the puzzle.

To better understand what is the actual issue, could you elaborate more on the total sequence of events, the observed behavior and the expected behavior?
To fix an issue it is important to understand the issue and this helps to reproduce it, for instance in a unit test, so it can be fixed.

hakehuang, 2022-08-03:
@ssharks , this is a iwl max-pro testing report, it belongs to TCP setup testing, which set the unused field in Destintion unreachable to a non-zero value.   I think the failure is at 
RFC 1122: section 4.2.3.9 {Dest. Unreach (2-4) => abort conn}