zephyrproject-rtos/test_results #1124: IPv6 Fragment Fragment: 512 MTU. Order: 1 2 3 ... N. Overlap: None. error

hakehuang, 2022-04-07:
**Describe the bug**
Fragment: 512 MTU. Order: 1 2 3 ... N. Overlap: None. test is Fail on Zephyr3.0.0 on qemu_x86

**References**

RFC 2460: section 4.5

**Results**

FAIL: icmp.v6 got no echo response, which is not an expected result.; PASS: udp.v6 got echo response, which is an expected result.; PASS: tcp.v6 got echo response, which is an expected result.

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: Zephyr3.0.0

hakehuang, 2022-04-08:
Also fails on qemu_x86 for Zephyr3.0.0

rlubos, 2023-03-15:
@hakehuang Any chance for a wireshark `pcap` from this test? I'd like to understand better on what are we dealing with here.

> FAIL: icmp.v6 got no echo response, which is not an expected result.; PASS: udp.v6 got echo response, which is an expected result.; PASS: tcp.v6 got echo response, which is an expected result.

Do I interpret it correctly, that **only** ICMPv6 fragmentation failed? And it worked fine for UDP/TCP?

rlubos, 2023-03-17:
@hakehuang I also need to understand what does `512 MTU` mean in terms of this test and others. At first I thought it's the underlying link MTU size, but now I'm not so sure about it, as it'd make little sense - the minimum MTU for IPv6 is 1280 bytes.

hakehuang, 2023-03-18:
> @hakehuang I also need to understand what does `512 MTU` mean in terms of this test and others. At first I thought it's the underlying link MTU size, but now I'm not so sure about it, as it'd make little sense - the minimum MTU for IPv6 is 1280 bytes.

good point, the MAX pro testing design so for the https://www.rfc-editor.org/rfc/rfc2460#section-4.5, but I do not know why they set the MTU to 512, let me check with IWL for help. Thanks a lot to point this out. @rlubos

hakehuang, 2026-10-30:
Closing as not planned for the current report, as requested when synchronizing issues to `v4.5.0-rc3-31-gd1d74eba22d` on `mimxrt1170_evk_cm7_1g`.

`IPv6.Fragment.003` is defined in the workbook's mapping sheet but has no rows in the supplied `test_results.csv`. This is a coverage/scope closure, not evidence that the defect is fixed or that this test passed. Reopen if the test is run and fails again.

closed 2026-10-30 by hakehuang