zephyrproject-rtos/test_results #1299: TCP Connected Change the window size to 1 fail

hakehuang, 2022-07-11:
Describe the bug
This test changes the window size field in all TCP packets to 1. This test is intended to exercise boundary conditions on the window size field.	Assumes the DUT is sending a TCP data stream to a data sink. DUT must be able to complete transmission of its reply in under 60 seconds.	The DUT must throttle back its transmissions to only one-octet-long data segments.	

References

RFC 793: section 3.7
RFC 1122: section 4.2.2.16 {Robust against shrinking window}

Results

FAIL: tcp.v4 got fewer bytes than expected, which is not an expected result.

Environment (please complete the following information):

OS: (e.g. Linux )
Toolchain (e.g Zephyr SDK)
Commit SHA or Version used: Zephyr3.1.0-1882-gb161554ee8

hakehuang, 2022-07-11:
bisect found the issue is introduced at 06ea11deffb73867e978c65247e785024f3a3811

dleach02, 2022-07-12:
The corrected bisect is: 74bc876bf5c1d860be62d318593991eecc6a6aeb. This is associated with https://github.com/zephyrproject-rtos/zephyr/pull/43018